esp-rs/esp8266-hal #8: Move memory.x into the HAL?

MabezDev, 2020-09-14:
Due to the fact the `esp32-hal` is only targeting one chip and not a family of chips, it made sense to include the memory.x.

Is there any reason not to do the same in this HAL?

jessebraham, 2020-09-14:
I believe all variants share the same memory map so I agree it would make sense to do so. I'm not certain on this, however.

icewind1991, 2020-09-16:
While all models share the same memory map afaik, I believe it still makes sense to split the cpu specific logic (exception/interrupt vectors, entry point, etc) from the peripheral logic of the esp8266(-hal) crates, which is why I put it in the xtensa crate.
Splitting the linker script as the esp32 crates does is an option but I don't really see an advantage for doing that here to justify the added complexity.

Note that this entire linker script thing is me only barely knowing what I'm doing and I might be missing something entirely.

MabezDev, 2020-09-19:
> I believe it still makes sense to split the cpu specific logic (exception/interrupt vectors, entry point, etc)

I agree, the memory.x file should only describe the memory layout of the chip, stuff like flash/sram sections.

> Splitting the linker script as the esp32 crates does is an option but I don't really see an advantage for doing that here to justify the added complexity.

It's probably fine for now, we only did it in the esp32 crates to be more inline with the other embedded rust pac crates. The only real use case for doing so is being able to use interrupts without the HAL, but that's not actually possible, at least on the esp32 as the interrupt matrix is a peripheral not part of the xtensa isa.

I see that memory.x has just been added, I'd say leaving the interrupt handlers in is fine for now, if someone needs that functionality we can always add it later.

closed 2020-09-19 by MabezDev