esp-rs/esp8266-hal #9: Build fails

alexandruradovici, 2020-09-24:
I am trying to build this, I used:

```bash
alex@alex-PC-i5:~/projects/esp8266-hal$ cargo build
   Compiling compiler_builtins v0.1.35
   Compiling core v0.0.0 (/home/alex/projects/rust-xtensa/build/x86_64-unknown-linux-gnu/stage1/lib/rustlib/src/rust/library/core)
   Compiling proc-macro2 v1.0.21
   Compiling unicode-xid v0.2.1
   Compiling syn v1.0.41
   Compiling semver-parser v0.7.0
   Compiling strsim v0.9.3
   Compiling fnv v1.0.7
   Compiling xtensa-lx v0.3.0
   Compiling ident_case v1.0.1
   Compiling xtensa-lx-rt v0.5.0
   Compiling esp8266-hal v0.4.0 (/home/alex/projects/esp8266-hal)
   Compiling paste v1.0.1
   Compiling semver v0.9.0
   Compiling rustc_version v0.2.3
   Compiling quote v1.0.7
   Compiling bare-metal v0.2.5
   Compiling darling_core v0.10.2
   Compiling rustc-std-workspace-core v1.99.0 (/home/alex/projects/rust-xtensa/build/x86_64-unknown-linux-gnu/stage1/lib/rustlib/src/rust/library/rustc-std-workspace-core)
   Compiling xtensa-lx-rt-proc-macros v0.1.0
   Compiling darling_macro v0.10.2
   Compiling alloc v0.0.0 (/home/alex/projects/rust-xtensa/build/x86_64-unknown-linux-gnu/stage1/lib/rustlib/src/rust/library/alloc)
LLVM ERROR: Error while trying to spill A10 from class AR: Cannot scavenge register without an emergency spill slot!
error: could not compile `compiler_builtins`

To learn more, run the command again with --verbose.
warning: build failed, waiting for other jobs to finish...
error: build failed
```

I installed the rust toolchain (following the quick start tutorial) and the esp toolchain. The esp32-hal is building fine.

jessebraham, 2020-09-24:
I'm not sure what the root of this problem is (sorry) but I believe it stems from LLVM. In Matrix some people found that building with `--release` seems to fix this. This obviously isn't an ideal solution but it may be enough to get you going for now. Are you able to confirm this is the case?

alexandruradovici, 2020-09-24:
Thank you. Yes, it works with `--release`.

closed 2021-04-16 by icewind1991