esp-rs/esp8266-hal #10: More examples

ahmtcn123, 2020-10-12:
Can we get Analog or I2S examples?

jessebraham, 2020-10-12:
These peripherals have yet to be implemented. Once they have been then examples can be added.

tsarquis88, 2021-11-01:
@jessebraham Is analog HAL coming soon?

jessebraham, 2021-11-02:
I unfortunately don't have a lot of time for this project right now as my focus has been on the newer chips, but I can try to find some time/energy to do it at some point.

Just as a bit of a background, I had made a previous attempt to implement this but was trying to do a pure-Rust implementation via reverse engineering. This became tedious and I gave up.

You can take a look at [this gist](https://gist.github.com/jessebraham/9b7ddcfa12f2d9d77de9dacc4c2ec9b3) so see an idea of how this implementation should look. I think at this point, the easiest path forward would be to provide the `system_adc_read` and `system_get_vdd33` functions from the ROM via a linker script and just call them directly.

tsarquis88, 2021-11-02:
Thanks for your job @jessebraham.
I'm a newbie in Rust so it's probably not gonna be me who solve this... Unless there's some docs about that kind of procedures.