esp-rs/esp8266-hal #20: Problem building package as a dependency

aghoneim92, 2022-04-08:
I'm getting an error when including this package as a dependency in another project. Building examples from the repo itself works fine, but when creating my own repo and adding `esp8622-hal = "0.5.0"` to dependencies, I get the following error while compiling `compiler_builtins`:
```
LLVM ERROR: Error while trying to spill A12 from class AR: Cannot scavenge register without an emergency spill slot!
```

jessebraham, 2022-04-08:
This is unfortunately an ongoing issue, you can see some context in this issue: https://github.com/esp-rs/rust/issues/87

I have updated the `master` branch of the repository semi-recently but have yet to release a new version. Not sure if this will make a difference or not for this specific bug, but it should at least allow building with the newest compiler.

For now I would recommend trying:

1. Install the most recent version of the Rust compiler fork with Xtensa support
    - There are pre-built artifacts and installation scripts available in the [esp-rs/rust-build](https://github.com/esp-rs/rust-build) repository
2. Change your `esp8266-hal` dependency to use git instead of a version number
    - eg.) `esp8266-hal = { git = "https://github.com/esp-rs/esp8266-hal" }`

I'm not sure if this will help or not, but it might. @MabezDev may be able to provide some more information as well.

aghoneim92, 2022-04-08:
Thanks for the quick reply! I was able to fix it by adding these parts of the `Cargo.toml` file from this repo:

```

# Place program completely in RAM (needed when e.g. using only ROM bootloader,
# or for debugging).
all_in_ram = []

[profile.dev]
lto = true
opt-level = 1

# Workaround for linker errors when using lto with the dev profile:
# https://github.com/rust-lang/compiler-builtins/issues/347
[profile.dev.package.compiler_builtins]
overflow-checks = false

[profile.release]
lto = true
```

MabezDev, 2022-04-10:
We actually just fixed (hopefully) this internally! It will be available in the next llvm release, at which point we will cut 1.60.0.1 of our custom Rust compiler :).

MabezDev, 2022-04-19:
Upgrade your compiler to 1.60.0.1 and this problem should go away :)

closed 2022-04-19 by MabezDev