esp-rs/esp8266-hal #21: Issue with building, error in xtensa-lx and xtensa-lx-rt crates

Squirrelcoding, 2022-05-28:
I was trying to build the crate for running the example but instead, I was met with some [errors](https://pastebin.com/691TjpEC). The errors are saying that the `asm!` and `global_asm!` macros aren't in the scope. I tried compiling this with the `nightly-x86_64-unknown-linux-gnu` toolchain, the `esp` toolchain, and the `stable-x86_64-unknown-linux-gnu` toolchain. How do I get rid of these errors so I can compile the crate?

jessebraham, 2022-05-30:
Unfortunately this project is mostly passively-maintained these days, so the most recent release is quite outdated. If you could try the `main` branch, I did push some updates last week to hopefully resolve any issues. If you can confirm this works for you I can cut a new release this week.

jessebraham, 2022-07-18:
All build errors should be resolved now with #22, but please feel free to re-open this if you experience any further troubles.

closed 2022-07-18 by jessebraham